catsystems/cats-embedded #4: Fix compiler warnings in cli.c

stojadin2701, 2021-07-29:
```
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c: In function 'cliSet':
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c:516:20: warning: declaration of 'value' shadows a previous local [-Wshadow]
  516 |           uint32_t value = strtoul(eqptr, NULL, 10);
      |                    ^~~~~
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c:512:13: note: shadowed declaration is here
  512 |     int16_t value = 0;
      |             ^~~~~
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c:523:15: warning: declaration of 'value' shadows a previous local [-Wshadow]
  523 |           int value = atoi(eqptr);
      |               ^~~~~
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c:512:13: note: shadowed declaration is here
  512 |     int16_t value = 0;
      |             ^~~~~
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c: In function 'cliPrint':
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c:613:6: warning: value computed is not used [-Wunused-value]
  613 |      *str--;
      |      ^~~~~~
At top level:
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c:651:13: warning: 'cliDefaultPrintLinef' defined but not used [-Wunused-function]
  651 | static bool cliDefaultPrintLinef(bool equalsDefault, const char *format, ...) {
      |             ^~~~~~~~~~~~~~~~~~~~
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c:638:13: warning: 'cliDumpPrintLinef' defined but not used [-Wunused-function]
  638 | static bool cliDumpPrintLinef(bool equalsDefault, const char *format, ...) {
      |             ^~~~~~~~~~~~~~~~~
C:\CATS\cats-embedded\boards\cats_rev1Pro\Core\Src\cli\cli.c:627:13: warning: 'cliPrintHashLine' defined but not used [-Wunused-function]
  627 | static void cliPrintHashLine(const char *str) {
      |             ^~~~~~~~~~~~~~~~
```

l-jost, 2021-08-08:
Add features and maybe split into multiple files

stojadin2701, 2021-12-31:
Closed by https://github.com/catsystems/cats-embedded/pull/61

closed 2021-12-31 by stojadin2701