catsystems/cats-embedded #66: Optimize Median Filter to O(n)

jbinz, 2021-11-27:


obalcells, 2021-11-28:
As the input data is quite small (9 values) it's not really worth it to use some fancy algorithm. I've tested it with this fancy algorithm (http://www.stat.cmu.edu/~ryantibs/papers/median.pdf) but it's just much slower (10x) than the current implementation for small values.

Because of this we just have the common and more basic algorithms to try out.

The most obvious choice from here is quickselect. quickselect is well-known and it's a derivation of quicksort. Basically we partition the array at each iteration but instead of recursing on both sides as we do with quicksort, we just go to the side where we know the median must be. We can do this because we know the rank of our pivot element. And you also don't need to merge sorted partitions because you're not sorting. However, it's still O(n^2) worst-case complexity so not linear ): But it doesn't really matter since the array is very small. The other (bigger) problem is that we must make a copy of the array. See details of the algorithm [here](http://www.foo.be/docs-free/Numerical_Recipe_In_C/c8-5.pdf).

I thought the main bottleneck was having to make a copy so I looked around for algorithms where you don't have to modify the original array. There's an algorithm called Torsten which does exactly this (implementation [here](http://ndevilla.free.fr/median/median/src/torben.c)). But unfortunately it's slower than quickselect.

Since the array is always of size 9 we could make something much simpler and just sort of brute-force the problem with if statements. But the code would probably look horrible. Fortunately I found [this](https://web.archive.org/web/20060613213236/https://www.xilinx.com/xcell/xl23/xl23_16.pdf) instead! Which is the same idea of hardwiring the code but it's faster and not horrible. The problem is that it only works if the array is of size 9 and you still have to make a copy of the array.

I've benchmarked all the algorithms with input size n = 9.

| Algorithm         | Mean execution time per 100 million calls | Std Dev |
|-------------------|-------------------------------------------|---------|
| Torben            | 9.93s                                     | 1.80s   |
| Quickselect       | 6.66s                                     | 1.21s   |
| Hardwired         | 3.35s                                     | 0.57s   |
| Sorting (current) | 34.87s                                    | 1.32s   |

Unsurprisingly, the hardwired algorithm is the fastest. What I would do is have the hardwired algorithm if n = 9 and quickselect otherwise.

obalcells, 2021-11-28:
Here's the code for the hardwired algorithm
```
#define SWAP(a,b) { float temp=(a);(a)=(b);(b)=temp; }
#define SORT_TWO(a,b) { if ((a)>(b)) SWAP((a),(b)); }

float median_hardwired(float array[], int array_size) {
    float tmp[array_size];
    memcpy(tmp, array, array_size * sizeof(float));

    SORT_TWO(tmp[1], tmp[2]); SORT_TWO(tmp[4], tmp[5]); SORT_TWO(tmp[7], tmp[8]);
    SORT_TWO(tmp[0], tmp[1]); SORT_TWO(tmp[3], tmp[4]); SORT_TWO(tmp[6], tmp[7]);
    SORT_TWO(tmp[1], tmp[2]); SORT_TWO(tmp[4], tmp[5]); SORT_TWO(tmp[7], tmp[8]);
    SORT_TWO(tmp[0], tmp[3]); SORT_TWO(tmp[5], tmp[8]); SORT_TWO(tmp[4], tmp[7]);
    SORT_TWO(tmp[3], tmp[6]); SORT_TWO(tmp[1], tmp[4]); SORT_TWO(tmp[2], tmp[5]);
    SORT_TWO(tmp[4], tmp[7]); SORT_TWO(tmp[4], tmp[2]); SORT_TWO(tmp[6], tmp[4]);
    SORT_TWO(tmp[4], tmp[2]);

    return(tmp[4]);
}
```

Here's quickselect
```
float quick_select(float input_array[], int array_size) {
    float array[array_size];
    memcpy(array, input_array, array_size * sizeof(float));
    
    float a;
    int i, j, m, k = (array_size >> 1);
    int l = 0, r = array_size - 1;

    while(r > l + 1) { // keep iterating until our partition is of size less than two
        m = (l + r) >> 1;
        assert(m > l && m < r);

        // we want array[l] <= array[l + 1] <= array[r]
        SWAP(array[m], array[l + 1]);
        SORT_TWO(array[l], array[r]);
        SORT_TWO(array[l + 1], array[r]);
        SORT_TWO(array[l], array[l + 1]);

        i = l + 1;
        j = r;
        a = array[l + 1];

        while(1) {
            i++; j--;
            while(array[i] < a) i++;
            while(array[j] > a) j--;
            if(j < i) break;
            SWAP(array[i], array[j]);
        }

        array[l + 1] = array[j];
        array[j] = a;

        // the new partition [l, r] must contain the median 
        if(j >= k) r = j - 1;
        if(j <= k) l = i; 
    }

    // we have the final partition which is of size 1 or 2
    if(r == l + 1) {
        SORT_TWO(array[l], array[r]);
    }

    return array[k];
}
```

obalcells, 2021-11-28:
Here's the code I've used to benchmark the different algorithms https://gist.github.com/obalcells/08de141c405d63c4e4f0e389c7f36b1c

stojadin2701, 2021-11-30:
Wow, 10x increase looks very promising ^^

Can you see if something like this could also work: https://github.com/regexident/median

obalcells, 2021-12-02:
It seems pretty cool, but it's definitely slower than the hardwired algorithm and it's in rust, so it would take some time to port it and test it. The hardwired algorithm is, I believe, the theoretical minimum of operations that you would need if the array size was 9. Therefore I would suggest to keep the hardwired algorithm and quickselect and if, for whatever reason, the median filter size gets increased, come back to this other algorithm.

stojadin2701, 2021-12-02:
Sounds good @obalcells, you can then submit a pull request with the hardwired approach and @jbinz will test it on the board  ^^

PR #69 (2021-12-04): Improved median filtering algorithm
This pull request fixes #66. I haven't managed to build the codebase myself yet, because I'm getting some errors with CMake, and I don't have enough time to resolve this problem until the weekend. So there could be some stupid compilation error. And the diff text looks weird, I don't know why.
closed 2021-12-04 by jbinz