catsystems/cats-embedded #333: Inaccuracies with altitude estimation through barometric pressure

kostas1507, 2023-08-03:
I was looking at the way altitude is calculated using pressure and I noticed that a constant temperature of 15C is used which I think can lead to significant inaccuracies in altitude estimation, especially at apogee. Let me explain:
The formula used is the hypsometric formula that normally takes in account current atmospheric temperature. At 3000M, it is not hard to imagine temperature dropping to -10C meaning that the formula using a set temperature of 15C can report an altitude of as much as 250M higher that the real altitude.
I understand it is really difficult to get accurate atmospheric temperature measurements during flight so I propose a solution that, I think, can greatly improve accuracy: get a single temperature measurement on the ground and substitute, in the hypsometric formula, the expected temperature according to the altitude (given that there is a ~2 degree drop every 300 meters). If I use T = T_0 - h/150, where h is the altitude and T_0 is the initial temperature, and solve for h in the hypsometric formula, I get h = (150*((P_0/P)^(-5.257)-1))/((P_0/P)^(-5.257)+149).
If I am correct, this should greatly improve accuracy and should be fixed. Our team is obligated to use the CATS flight computer for EUROC and the accuracy of our altitude on apogee is assessed using this software. We are preparing a kalman filter to estimate altitude during ascent for use with our airbrakes and it would be assuring to know that altitude estimation by CATS is as accurate as it can be.
I am would be happy to make the necessary fixes in the code

kostas1507, 2023-08-03:
an even simpler solution that will still improve accuracy a lot is only changing the formula and assuming that the starting temperature is 15C. This might be a more robust solution as temperature inside the avionics tube might be much higher than outside temperature!

l-jost, 2023-08-03:
Hello @kostas1507 
We are using the [International Standard Atmosphere](https://en.wikipedia.org/wiki/International_Standard_Atmosphere). This model creates a common reference and all teams should use it for their calculations. The comment in the firmware needs to be updated though :)

closed 2023-08-03 by l-jost

kostas1507, 2023-08-03:
Thanks for clarifying, I used the formula provided in [this](url:https://keisan.casio.com/exec/system/1224579725) website that even provides [proof](url:https://keisan.casio.com/keisan/image/Convertpressure.pdf) of their formula but after reading the wikipedia page you sent I think they have made a mistake in their calculations

l-jost, 2023-08-03:
I unit tested the calculation against this [table ](https://onlinelibrary.wiley.com/doi/pdf/10.1002/9781118534786.app1) and everything seems fine.

kostas1507, 2023-08-04:
Looks like it! sorry the link didn't work last time, this was the website I used: https://keisan.casio.com/exec/system/1224585971 . The calculator it provides takes in current temperature and does not confirm the table you send if I use 15C. But If I use the formula provided in the website, I do need to set the temperature to 15C to confirm the table. I guess they are using a different formula than in the calculator without disclosing it. Sorry for that..